Buckitup-chat/chat #65: Multiple download. Files only mode

sergey-lukianov, 2022-12-16:
We want to introduce downloading multiple files without `zip`ing  them.
As shown in first video here. -https://github.com/Buckitup-chat/chat/pull/61

In case if only images, videos and files are selected. Any other type should trigger zipped download.

closed 2022-12-22 by sergey-lukianov

buckitupnet, 2023-01-06:
Is it possible to approve the download just once instead for every single file?

almirsarajcic, 2023-01-06:
We don't have control of that as it's the browser that's giving you the prompt. Only when we download the archive we can have only one warning, because it's only one file.

https://elixirforum.com/t/best-way-to-send-multiple-files-as-http-response/17768/3
https://stackoverflow.com/a/1041591/920303

PR #68 (2022-12-22): Download multiple media files without archiving
Quick solution for downloading multiple files without zipping, in case the selected messages are files, images, and/or videos.
A more robust solution might be required in which case I'll need to refactor this. That would involve more client-to-server communication.

Resolves #65.