Buckitup-chat/chat #67: Global Uploader

sergey-lukianov, 2022-12-16:
The main goal is to detach uploads from chat/room to be able to switch them while uploading.
When upload completes it gets added to chat/room it started in.
Uploads are getting send in window of 2 max. I.e. if we uploading 11 files, 2 files are being send and others are awaiting their turn.

We may add cancel and pause to each upload.

Extra feature: be able to resume upload if connection dropped or client gets suspended.

## Tech design
Uploads tracking should be bound to the LiveVew. It should know what chat/room each upload designates.
List of uploads should be implemented as a component. Upload button as a component as well.

To track resumable uploads we may additionally store hash of 
 - client file path
 - client file size
 - destination chat/room 
 Having this we may recognise previously (partially) uploaded file. 
If partially uploaded file was not completed for a day we should delete it. To do so we need to track last file upload attempt timestamp and current client time.
As the platform has no time source, we may rely only on client clock. (Each client may have set his clock to different time)

## Design guidelines
On mobile  it should be 2 options 
 - upload a file 
 - upload an image or video (image/*, video/*)

On mobile upload screen may be triggered by upload  button, having file and image upload options in bottom and uploads list above them.

On desktop it might be 
 1. Button stays same. Uploader list goes as chat or room list
 2. Uploader button makes bottom half of chat/room list to become uploader list with `add new upload` option

buckitupnet, 2022-12-17:
[image]

almirsarajcic, 2022-12-24:
Sorry @sergey-lukianov, I forgot to ask, why do we need 2 buttons for file upload? I guess we could use client file type for that.

Also, what do you think about using file content hash (reading on the client-side using fileReader) and/or `lastModified` to compare with the previously failed uploads?

sergey-lukianov, 2022-12-26:
@almirsarajcic  2 buttons on mobile
 - general one for any files
 - media one with `accept: video/*, image/*, audio/*` makes Android (at least) to show gallery
 
reading hash of 2+ Gb video might take time on the client. If it is fast (less than 1 sec) - let's do it

PR #93 (2023-01-14): File upload improvements (Global Uploader)
Resolves #67.

- [x] Simultaneous uploads in multiple dialogs/rooms
- [x] Max 2 concurrent uploads (user can override by manually resuming paused uploads)
- [x] Pausing, resuming and canceling uploads
- [x] 2 upload buttons on Android
- [x] Resuming failed uploads
- [x] Pruning stale uploads
- [x] Improved design
closed 2023-01-14 by sergey-lukianov