Buckitup-chat/chat #96: The smart sync:

buckitupnet, 2023-01-06:
Only custom content has to be synchronized during the backup.
Modes:
 - Normal. Everything is synced (like currently implemented)
 - Onliners. Only content accessible to the users online is synchronized
 - Cargo. Only the content of a particular room syncronized.

sergey-lukianov, 2023-01-14:
Initial bit of work is in #113 

We have Supervision tree for each DB. Internal DB(SD card) is handled by `chat`.
Main DB (first USB plugged) and Backup(2nd USB plugged) are handled by `platform`.
The idea is to tweak BackupDB supervisor that after mounting it could decide which functionality (feature Supervision tree) to apply.
This can be done by checking directory structure of flash drive
* `bdb` - backup DB
* `cargo_db` - cargo DB w/ truck room 
If none of this found we are checking a setting, which can be set in Admin room to Backup/Onliners/Cargo/etc

### Onliners
This should create DB in `main_db` folder and copy into newly created DB info that current online users can access.
This might work like this
1. Send pubsub request for pub_keys hashes. Each user will send it own pub_key as well as all rooms it have
2. Key_scope algo goes thru DB checking which keys to copy. Returning read stream
3. Read_stream is set in newly created DB WriteQueue to start copying.
4. After finishing data copy we need to signal to user in some way.

### Cargo
Cargo DB should contain the only room and chats with invitations into the room for all checkpoints.

1. Checkpoint will join the room via chat for it.
2. Will read sensor info into the room
3. Signal in some way about completion.  

Docs should cover DB supervision, DB syncing and Feature Supervision proposal (Plug for OTP)

sergey-lukianov, 2023-01-19:
Docs: 
* https://github.com/Buckitup-chat/chat/blob/main/lib/chat/db/prioritization.livemd
* https://github.com/Buckitup-chat/chat/blob/main/lib/chat/db/supervision.livemd

Following section is tech design for this issue - https://github.com/Buckitup-chat/chat/blob/main/lib/chat/db/supervision.livemd#media-drive-supervision-proposal

closed 2023-03-08 by sergey-lukianov