Buckitup-chat/chat #101: usb drive dump

buckitupnet, 2023-01-10:
Another  sync mode & room type - when activated (created?) all data from external usb drive when plugged (e.g. from a camera or a drone) will be sucked automatically as a sequence of files to the dedicated room.
The process has to be indicated in the "drive tray"

sergey-lukianov, 2023-01-12:
@buckitupnet 
1. We need to keep the key of such room somewhere (admin settings?).
2. Should this room be the only one for all RPis or one per RPi?
3. Should take part in backup?
4. How user will enter a room?

buckitupnet, 2023-01-12:
1. yes, admin database seems the right place. the user who creates the room has the key also as normal, can invite and so on.
2. this room may be synced across the devices  as usual, but the key remains only where created, so when sinked it changes the type to regular private room.
3. yes (see above)
4. admin always can enter send invites and accept requests. The creator also stores the key.

sergey-lukianov, 2023-01-13:
### Drive dump room

Drive dump room can be created when there is no 2nd USB drive inserted (backups and dumps finished) and there is no active drive room.
It would be one more option in room creation dialog. 
Upon creation it 
 - creates private(request) room
 - register the room as DriveDump (in DriveDump GenServer)
 - new indication appears near by db icon (awaiting 5 mins for usb drive to be plugged)

On device plug we
1. check if folder structure is like a camera has (`DCIM` like) and DriveDump room is registered
2. get pub_key of the room (DriveDump GenServer)
3. load all files into the room as if it were uploaded by room creator (using room.admin_hash as author_hash)
4. on load completion blink leds and remove indication from sidebar

buckitupnet, 2023-01-14:
Dump lock: in some cases we want the dump state to persist without a client setting it up manually for every usb drive (when in rush, field conditions etc..). While default dump room deactivates when one usb dump is done and/or 5 minutes of inactivity.  In the locked scenario the dump room persists until unlocked by the user (or admin). Until than no other dump-rooms can be created.
[image]
[image]
[image]

sergey-lukianov, 2023-04-05:
Files from drive should be stored in order of file creation

File (image/video) messages should have same timestamp as file timestamp on drive

closed 2023-04-15 by sergey-lukianov