Buckitup-chat/chat #114: Public room post permanent link

buckitupnet, 2023-01-15:
In a public room - in the context menu for a post one more option has to be added: "link", when prssed:
1. the room key is stored in the Admin database and hosted at some ulr for anyone to be able to join (even when first time on the platform and nobody is online to approve the request)
2. the particular post "address" is linked to the ulr such that when followed the link any (even first time login, going through the registration screen in this case) gets directly to the intended room and the particular post  in it. 
3. multiple links may be created likewise in one room leading to different posts
4. when the feature is in use (the room key is hosted publicly) "linked"  notification shows up in the room header with the "(unlink?)" button. When "unlink" is triggered (with confirmation popup), the room key is erased from the host and all links previously created, cancel out.
[image]
[image]

sergey-lukianov, 2023-01-17:
1. Why should we have it in AdminDB? Room should not be copied or it should always be present on device?
6. When room is unlinked, where should earlier created link lead?

buckitupnet, 2023-01-18:
1q. the link isn't global, and the priv. key shouldn't go travel somewhere...
6q. the link will stop working (probably 404 page may be created)

Optionally we may consider encoding the key itself to the qr-code?

closed 2023-04-19 by sergey-lukianov