Buckitup-chat/chat #118: removing the user kills his room

buckitupnet, 2023-01-17:
When a user  removed (by the admin tool), the room previously created by him isn't accessible any more by another user (joined earlier).
How architecturally inevitable is this? )

sergey-lukianov, 2023-01-17:
It is likely to raise a rendering error or in signature check. the Log would help

buckitupnet, 2023-01-18:
Couldn't reproduce, sorry, will update with the log if see it again.

sergey-lukianov, 2023-01-20:
This likely to be first room creation fail
The log is 
```
web.1  | 10:50:51.385 [error] GenServer #PID<0.793.0> terminating
 web.1  |     (chat 0.1.0) lib/chat_web/live/main_live/page/room.ex:43: ChatWeb.MainLive.Page.Room.init/2
 web.1  | ** (KeyError) key :pub_key not found in: nil. If you are using the dot syntax, such as map.field, make sure the left-hand side of the dot is a map
 web.1  |     (chat 0.1.0) lib/chat_web/live/main_live/index.ex:273: ChatWeb.MainLive.Index.handle_event/3
 web.1  |     (phoenix_live_view 0.18.3) lib/phoenix_live_view/channel.ex:382: anonymous fn/3 in Phoenix.LiveView.Channel.view_handle_event/3
 web.1  |     (stdlib 4.1.1) proc_lib.erl:250: :proc_lib.wake_up/3
 web.1  |     (telemetry 1.1.0) /tmp/build/deps/telemetry/src/telemetry.erl:320: :telemetry.span/3
 web.1  |     (phoenix_live_view 0.18.3) lib/phoenix_live_view/channel.ex:216: Phoenix.LiveView.Channel.handle_info/2
 web.1  |     (stdlib 4.1.1) gen_server.erl:1123: :gen_server.try_dispatch/4
 web.1  |     (stdlib 4.1.1) gen_server.erl:1200: :gen_server.handle_msg/6
 web.1  | Last message: %Phoenix.Socket.Message{topic: "lv:phx-Fzv-xNHKo9hcZAXB", event: "event", payload: %{"event" => "room/create", "type" => "form", "value" => "new_room%5Bname%5D=Open+Safari+%231&new_room%5Btype%5D=public"}, ref: "12", join_ref: "4"}
 web.1  | 10:50:52.366 [warning] [db writer] fsyncing Chat.Db.InternalDb
```

PR #133 (2023-01-21): db unleash: room creation do not wait for DB
Resolves #118
closed 2023-01-21 by sergey-lukianov