Buckitup-chat/chat #119: Uploading the same file again causes an error

almirsarajcic, 2023-01-18:
When I upload the same file of any type twice and try to reopen the dialog/room it shows an error

https://user-images.githubusercontent.com/845364/213134383-daac995d-e79e-4f72-a3e2-74bad83372fc.mov

```
[error] GenServer #PID<0.978.0> terminating
** (Jason.EncodeError) invalid byte 0x8C in <<111, 140, 40, 55, 229, 165, 141, 177, 118, 113, 190, 48>>
    (jason 1.4.0) lib/jason.ex:213: Jason.encode_to_iodata!/2
    (phoenix 1.7.0-rc.0) lib/phoenix/socket/serializers/v2_json_serializer.ex:72: Phoenix.Socket.V2.JSONSerializer.encode!/1
    (phoenix 1.7.0-rc.0) lib/phoenix/channel/server.ex:252: Phoenix.Channel.Server.reply/6
    (phoenix_live_view 0.18.3) lib/phoenix_live_view/channel.ex:825: Phoenix.LiveView.Channel.reply/4
    (phoenix_live_view 0.18.3) lib/phoenix_live_view/channel.ex:671: Phoenix.LiveView.Channel.handle_changed/4
    (stdlib 4.1.1) gen_server.erl:1123: :gen_server.try_dispatch/4
    (stdlib 4.1.1) gen_server.erl:1200: :gen_server.handle_msg/6
    (stdlib 4.1.1) proc_lib.erl:240: :proc_lib.init_p_do_apply/3
Last message: %Phoenix.Socket.Message{topic: "lv:phx-FztdFfkonHMHvg2E", event: "event", payload: %{"event" => "dialog/switch", "type" => "click", "value" => %{"user-id" => "a69965152df455939f068a2ba3a1c968453eec66f1137da0518137c54c644a7e", "value" => 0}}, ref: "8", join_ref: "4"}
State: %{components: {%{}, %{}, 1}, join_ref: "4", serializer: Phoenix.Socket.V2.JSONSerializer, socket: #Phoenix.LiveView.Socket<id: "phx-FztdFfkonHMHvg2E", endpoint: ChatWeb.Endpoint, view: ChatWeb.MainLive.Index, parent_pid: nil, root_pid: #PID<0.978.0>, router: ChatWeb.Router, assigns: %{__changed__: %{}, db_status: %{compacting: false, flags: [], mode: :internal, writable: :yes}, dialog: nil, flash: %{}, image_gallery: nil, is_admin: true, joined_rooms: [], live_action: :index, lobby_mode: :chats, locale: "en-GB", logout_step: nil, me: #Chat.Identity<name: "nesto", ...>, mode: :lobby, monotonic_offset: 2250494753, my_id: "a69965152df455939f068a2ba3a1c968453eec66f1137da0518137c54c644a7e", need_login: false, new_rooms: [], operating_system: "Mac OS X", peer: nil, room_map: %{"0a1d7cd41fc9af7726e77a724bb8fae8a75553cf73ad6604ef6b1d53e4f7f562" => #Chat.Identity<name: "", ...>, "0e762256266198f14e403afa048765b2140de8425d8236d6f16fb95a6b0cf0ea" => #Chat.Identity<name: "", ...>, "1fbc7e846d12636c617b7c6f4b6d1df059921dd2f67d2d261de4524f4fb7a40c" => #Chat.Identity<name: "", ...>, "33ed500178582da4b337c386b6b3480d998b8c667be38b0305e3fee6544049c0" => #Chat.Identity<name: "", ...>, "359c63bfcd18d5eb7ce5b6903d0bbcf104a09886c180398a5e2719fb53178cac" => #Chat.Identity<name: "", ...>, "39558f360a7c60e5e9991173b18c9873d27ab8975071ef3b4e6caba14a89531d" => #Chat.Identity<name: "", ...>, "3ed66bb6a0f2e9172cbf9f5b36d0a9a5885005c7782fe5d36cd9a2aebe2bb396" => #Chat.Identity<name: "", ...>, "5fb73c76b61981fea156fc96f1c74382550527bc9c86a1445b64dff0bdafaae3" => #Chat.Identity<name: "", ...>, "7772d00f760a8ac326da711dd3eea092072a014ebd2a2db418b0120a78384a16" => #Chat.Identity<name: "", ...>, "7afc629c78b717deff2dc61845589347808d373f742993b90938e563038628fb" => #Chat.Identity<name: "", ...>, "8863728be4489623b5ec504b72e03abd62cbe4c1358c022d2f3552cc0980efe6" => #Chat.Identity<name: "", ...>, "944ded67f9f0cbf7af54d1a7cd62276135bc96cb5e84598ea03eaba3bd5bdbe0" => #Chat.Identity<name: "", ...>, "b93dcbb2d8489013c081b78e6860e21c95865e9b6cc18ad99455a75efcc10490" => #Chat.Identity<name: "", ...>, "d00cf8148cfad2944381fb3edd55dc330ec5f823ce981590bd7ac5b211a12ca3" => #Chat.Identity<name: "", ...>, "de9f28cc63a8099efe99a1ef1cc297959441107dc7c1b4e3e9b71d4b40cfb026" => #Chat.Identity<name: "", ...>, "ecc05255ca80825388b76125b3935af21eb6fc4b54c0e4e0384a15d5c502e0b3" => #Chat.Identity<name: "", ...>, "f299819016aaa49ff96d5a0515c7a1c5282e82f469e841ad3ea529e2b5b5ec72" => #Chat.Identity<name: "", ...>}, rooms: [#Chat.Identity<name: "", ...>, #Chat.Identity<name: "", ...>, #Chat.Identity<name: "", ...>, #Chat.Identity<name: "", ...>, #Chat.Identity<name: "", ...>, #Chat.Identity<name: "", ...>, #Chat.Identity<name: "", ...>, #Chat.Identity<name: "", ...>, #Chat.Identity<name: "", ...>, #Chat.Identity<name: "", ...>, #Chat.Identity<name: "", ...>, #Chat.Identity<name: "", ...>, #Chat.Identity<name: "", ...>, #Chat.Identity<name: "", ...>, #Chat.Identity<name: "", ...>, #Chat.Identity<name: "", ...>, #Chat.Identity<name: "", ...>], timezone: "Europe/Sarajevo", timezone_offset: 1, uploads: %{__phoenix_refs_to_names__: %{"phx-FztdFgl974gxOg3E" => :file, "phx-FztdFgl9tPMxOg2k" => :my_keys_file}, file: #Phoenix.LiveView.UploadConfig<name: :file, max_entries: 2000, max_file_size: 102400000000, entries: [], accept: :any, ref: "phx-FztdFgl974gxOg3E", errors: [], auto_upload?: true, progress_event: #Function<3.93777003/3 in ChatWeb.MainLive.Index.handle_chunked_progress>, ...>, my_keys_file: #Phoenix.LiveView.UploadConfig<name: :my_keys_file, max_entries: 1, max_file_size: 1024000000, entries: [], accept: :any, ref: "phx-FztdFgl9tPMxOg2k", errors: [], auto_upload?: true, progress_event: #Function<1.93777003/3 in ChatWeb.MainLive.Index.handle_progress>, ...>}, uploads_metadata: %{}, user_id: #Chat.Card<name: "nesto", hash: "a69965152df455939f068a2ba3a1c968453eec66f1137da0518137c54c644a7e", ...>, users: [#Chat.Card<name: "nesto", hash: "a69965152df455939f068a2ba3a1c968453eec66f1137da0518137c54c644a7e", ...>], version: "version should be here"}, transport_pid: #PID<0.969.0>, ...>, topic: "lv:phx-FztdFfkonHMHvg2E", upload_names: %{}, upload_pids: %{}}
```

The issue got introduced in [d518446](https://github.com/Buckitup-chat/chat/commit/d5184464393180fd8258b5ed91ddb376669947e1). Here's the diff between the working and the current version: https://github.com/Buckitup-chat/chat/compare/uploader-working...uploader-error

I suspect it's because two files have the same upload key.

sergey-lukianov, 2023-01-19:
Do nothing on it till it gets discussed. I will update on results.
We might reuse previously uploaded file

sergey-lukianov, 2023-01-20:
We should not upload the file once again in same room/dialog. We should just link it to a new message.

Also we need to track to which messages we have linked the file. Do we could do `reference counting` and deletea file when last message linking to it gets deleted.
This can be done by adding message id  into `file_index`. I.e it become 
```elixir
{{:file_index, reader_hash, file_key, message_id},
 true}
```

Note: in  case of dialog message deletion we need to check links on both sides of dialog. 

Documentation and tests should reflect this change as well

almirsarajcic, 2023-01-25:
The cause of the error was that the `{:file, key}` was overwritten with the new secret
https://github.com/Buckitup-chat/chat/blob/ac45de464d2ff9483793710b0f3690b4784fe117/lib/chat/files.ex#L23-L28
Because of that, the file from the older message couldn't get accessed.

Fixed in https://github.com/Buckitup-chat/chat/pull/137/commits/5397a0960945385315da65be8d0653ccb7b0f90f by putting file data under a new key.

PR #137 (2023-01-27): Fix duplicate files error and skip uploading duplicate files
Fixes #119.
closed 2023-01-27 by sergey-lukianov