Buckitup-chat/chat #167: show the free space available for admin

buckitupnet, 2023-02-13:
Admin panel needs indication of the free space available on the drives where we can:
the main / internal / backaup (during the backup process).

sergey-lukianov, 2023-04-13:
Several drives can be used with the device.
Admin need a clear way to understand how much spcae left on each of devices.

RPi operates from SD card. It is our internal storage. It might be pretty limited.
First inserted USB drive treated as main. All the updates come to it. Even if internal is full already.
Second inserted device treated as media. It can be used for backup, partial backup, dump drive or cargo (special scenario).

We need a widget that will reflect free space in real time for those 3 devices (if inserted)
- intermal
- main
- media

Optionally, we can display if device is switched into read only mode (got full).

More technical details - https://github.com/Buckitup-chat/chat/blob/main/lib/chat/db/supervision.livemd
It could be something like
```elixir
[Chat.InternalDB, Chat.MainDB, Chat.MediaDB]
|> Stream.filter(&Process.alive/1)
|> Stream.map(&Chat.Maintenance.db_free_space/1)
|> Enum.to_list()
```

P.S. #100 duplicates this issue and should be closed on completion of this one

giangnt41, 2023-04-21:
Hi Sergey, sorry for taking so long to understand about raspberry pi and the project. 😅 

Please check #290 and tell me what you think 😃

closed 2023-04-27 by sergey-lukianov