Buckitup-chat/chat #227: Cargo checkpoints list

buckitupnet, 2023-03-22:
Admin panel need the new vidget for generating the invite-list, just one for the platform like the admins list.
Users may be added or removed to/from the cargo-list  by the Admin
When the Cargo scenario is active (switch in the Admin room is enabled) new option for the cargo room is available in the rooms-types list - "the cargo room"
When cargo room is created, invites automatically sent to all contacts in the cargo list.
When **naming the new cargo-room**, the platform makes sure that the name is **new** (unique). UI only lets only open the existing rooms, if the given name matches with any rooms on the system.  In this case the New room is not created (_invites are not sent_ and we may want to send invites separately?).
When a second, empty USB drive is plugged it is dedicated to the last cargo room created. (we need some ui confirmation of that..?)
The cargo sync performed - the dedicated room is synced, and all the dialogues of the users **from cargo-list** are, synced.

How we make new cargo-drive for the existing cargo-room?

buckitupnet, 2023-03-22:
Proposal:

- when cargo-sync is activated in Admins room -> no repeating names are allowed for rooms.
- cargo room is created normally just like any other room (given it's name must be unique) 
- any room may be triggers to become cargo room (for some limited time ~2 minutes) this can be canceled.
- in that case the invites are being send to all users on cargo-list, and the room becomes synced with this invites on the dedicated flash-drive, after it is plugged (see above)
- before it happens, the UI shows a modal dialogue "plug in the empty" flash drive (or "cancel"). platform waits for the cargo drive. If the drive has cargo room (or any other db), or time passes, or canceled by user the cargo room is not created. 
- if empty drive attached, the cargo-room creation completes and and it is assigned to that drive.
- the sync and invites to the list performed.
- when complete modal notification is flipped into "cargo-sync initiated successfully" , "ok".
- when a drive with existing cargo  room is plugged, the separate tab appears with that particular cargo room to be accessed directly.

buckitupnet, 2023-04-21:
Not closing this one since still the rooms list isn't updated properly upon sync on a new system.

almirsarajcic, 2023-04-25:
Fixed in https://github.com/Buckitup-chat/chat/pull/292.

closed 2023-04-26 by buckitupnet