Buckitup-chat/chat #273: Error displaying message

sergey-lukianov, 2023-04-15:
```
2023-04-14 14:03:34.634000 [error] GenServer #PID<0.3387.0> terminating
** (KeyError) key :url not found in: nil. If you are using the dot syntax, such as map.field, make sure the left-hand side of the dot is a map
    (chat 0.1.0) lib/chat_web/live/main_live/layout/message.ex:314: anonymous fn/2 in ChatWeb.MainLive.Layout.Message."message (overridable 1)"/1
    (chat 0.1.0) /Users/lukianov/work/buckitUp/chat/lib/chat_web/live/main_live/layout/message.ex:86: ChatWeb.MainLive.Layout.Message.message_block/1
    (elixir 1.14.1) lib/enum.ex:2468: Enum."-reduce/3-lists^foldl/2-0-"/3
    (phoenix_live_view 0.18.18) lib/phoenix_live_view/diff.ex:396: Phoenix.LiveView.Diff.traverse/7
    (phoenix_live_view 0.18.18) lib/phoenix_live_view/diff.ex:571: anonymous fn/3 in Phoenix.LiveView.Diff.traverse_comprehension/5
    (elixir 1.14.1) lib/enum.ex:1780: Enum."-map_reduce/3-lists^mapfoldl/2-0-"/3
Last message: {:room, {:new_message, {1, %Chat.Rooms.Message{timestamp: 1681481012, author_key: <<2, 251, 142, 52, 47, 8, 251, 107, 71, 225, 133, 164, 65, 39, 169, 97, 70, 51, 30, 238, 248, 55, 16, 67, 186, 219, 62, 227, 217, 128, 9, 71, 222>>, encrypted: {<<212, 35, 101, 205, 81, 183, 116, 17, 2, 21, 224, 90, 22, 3, 151, 171, 66, 10, 133, 174, 55, 233, 131, 132, 109, 254, 202, 24, 18, 29, 149, 80, 32, 30, 46, 82, 195, 78, 3, 129, ...>>, <<31, 125, 18, 121, 180, 146, 189, 88, 183, 28, 69, 182, 22, 207, 160, 60, 64, 131, 166, 66, 62, 232, 64, 87, 147, 85, 29, 169, 178, 172, 174, 113, 32, 119, 47, 198, 202, 223, 78, ...>>}, type: :video, id: "2b6f877a-ca03-40ec-84f8-9fef82a012bd", version: 1}}}}
State: %{components: {%{2 => {ChatWeb.MainLive.Page.RoomForm, :room_form, %{__changed__: %{}, changeset: #Ecto.Changeset<action: nil, changes: %{}, errors: [name: {"can't be blank", [validation: :required]}], data: #Chat.Rooms.RoomInput<>, valid?: false>, flash: %{}, id: :room_form, media_settings: %Chat.Admin.MediaSettings{functionality: :backup}, myself: %Phoenix.LiveComponent.CID{cid: 2}}, %{__changed__: %{}, lifecycle: %Phoenix.LiveView.Lifecycle{handle_event: [], handle_info: [], handle_params: [], mount: [], after_render: []}, root_view: ChatWeb.MainLive.Index}, {197259578132614148269409409101450331944, %{0 => {118970292807654719229453556541006727223, %{3 => {221662214080527960856883882372570928791, %{1 => {322878696178084776782415127383119283351, %{}}, 2 => {229987047553272858979023794365556468638, %{}}, 3 => {229987047553272858979023794365556468638, %{}}, 4 => {229987047553272858979023794365556468638, %{}}, 6 => {322878696178084776782415127383119283351, %{}}, 7 => {325983420552817819431408813306717919371, %{}}}}}}}}}, 3 => {ChatWeb.MainLive.Layout.ImageGallery, "imageGallery", %{__changed__: %{}, current: nil, flash: %{}, id: "imageGallery", is_open?: false, list: [], myself: %Phoenix.LiveComponent.CID{cid: 3}, next: nil, prev: nil, room_identity: #Chat.Identity<name: "rooom", ...>, type: "room"}, %{__changed__: %{}, lifecycle: %Phoenix.LiveView.Lifecycle{handle_event: [], handle_info: [], handle_params: [], mount: [], after_render: []}, root_view: ChatWeb.MainLive.Index}, {218034841835605869015792847686752404577, %{}}}}, %{ChatWeb.MainLive.Layout.ImageGallery => %{"imageGallery" => 3}, ChatWeb.MainLive.Page.RoomForm => %{room_form: 2}}, 4}, join_ref: "4", serializer: Phoenix.Socket.V2.JSONSerializer, socket: #Phoenix.LiveView.Socket<id: "phx-FiTV1vUvFOPFspJh", endpoint: ChatWeb.Endpoint, view: ChatWeb.MainLive.Index, parent_pid: nil, root_pid: #PID<0.3387.0>, router: ChatWeb.Router, assigns: %{timezone_offset: 3, my_id: <<2, 251, 142, 52, 47, 8, 251, 107, 71, 225, 133, 164, 65, 39, 169, 97, 70, 51, 30, 238, 248, 55, 16, 67, 186, 219, 62, 227, 217, 128, 9, 71, 222>>, lobby_mode: :rooms, db_status: %{compacting: false, flags: [], mode: :main, writable: :yes}, dialog: nil, message_update_mode: :replace, logout_step: nil, uploads_metadata: %{"0cf05713-ad52-4645-b741-8e5a6c277244" => %Chat.Upload.UploadMetadata{credentials: {<<47, 93, 194, 164, 59, 205, 149, 184, 64, 221, 162, 124, 161, 207, 14, 208, 8, 34, 205, 138, 5, 128, 122, 204, 111, 55, 227, 216, ...>>, <<29, 45, 40, 82, 104, 37, 236, 78, 112, 220, 233, 109, 179, 229, 155, 136, 87, 14, 116, 214, 238, 53, 165, 221, 4, 110, 151, ...>>}, status: :pending, destination: %{pub_key: "0309678f0ca5428d7885a106d1bdd1d587703a979601f73222071e16548f628852", type: :room}}, "1d673204-01f0-4cf9-aa42-516fc1e808c3" => %Chat.Upload.UploadMetadata{credentials: {<<241, 252, 95, 207, 209, 71, 104, 239, 233, 146, 126, 167, 52, 14, 142, 194, 47, 169, 6, 179, 26, 0, 86, 17, 106, 136, 232, ...>>, <<243, 61, 206, 214, 12, 152, 234, 209, 226, 243, 7, 49, 138, 146, 173, 150, 227, 75, 213, 161, 8, 116, 202, 183, 22, 133, ...>>}, status: :active, destination: %{pub_key: "0309678f0ca5428d7885a106d1bdd1d587703a979601f73222071e16548f628852", type: :room}}, "1dd38101-cb20-42d9-b2cf-c0d3aa09d5f6" => %Chat.Upload.UploadMetadata{credentials: {<<207, 109, 249, 212, 23, 152, 153, 33, 56, 9, 74, 91, 16, 158, 82, 114, 129, 74, 61, 26, 100, 173, 213, 127, 41, 155, ...>>, <<100, 1, 77, 162, 109, 181, 2, 127, 10, 1, 220, 6, 8, 149, 233, 94, 125, 196, 109, 116, 131, 12, 250, 209, 191, ...>>}, status: :pending, destination: %{pub_key: "0309678f0ca5428d7885a106d1bdd1d587703a979601f73222071e16548f628852", type: :room}}, "33fa91b5-9894-4362-ad30-e055723be10e" => %Chat.Upload.UploadMetadata{credentials: {<<35, 163, 209, 64, 101, 104, 130, 86, 213, 192, 217, 247, 60, 11, 217, 190, 109, 157, 209, 64, 229, 71, 5, 52, 26, ...>>, <<232, 140, 212, 140, 108, 144, 229, 48, 58, 250, 64, 157, 147, 31, 110, 202, 126, 165, 20, 167, 140, 175, 91, 106, ...>>}, status: :pending, destination: %{pub_key: "0309678f0ca5428d7885a106d1bdd1d587703a979601f73222071e16548f628852", type: :room}}, "42bac818-ffc0-4073-8b0e-da45094a8c5f" => %Chat.Upload.UploadMetadata{credentials: {<<126, 23, 179, 131, 171, 122, 75, 91, 176, 60, 185, 194, 96, 171, 74, 132, 31, 190, 75, 107, 154, 164, 29, 184, ...>>, <<103, 154, 238, 97, 120, 78, 181, 1, 195, 157, 66, 172, 120, 144, 222, 218, 206, 176, 93, 24, 245, 82, 205, ...>>}, status: :pending, destination: %{pub_key: "0309678f0ca5428d7885a106d1bdd1d587703a979601f73222071e16548f628852", type: :room}}, "4e172346-46b8-4f88-8230-4391a018e799" => %Chat.Upload.UploadMetadata{credentials: {<<10, 58, 104, 79, 176, 13, 59, 250, 66, 212, 8, 82, 163, 219, 8, 59, 46, 41, 56, 64, 169, 123, 221, ...>>, <<127, 232, 34, 66, 102, 128, 135, 100, 232, 180, 200, 148, 58, 200, 191, 156, 50, 21, 239, 99, 147, 212, ...>>}, status: :pending, destination: %{pub_key: "0309678f0ca5428d7885a106d1bdd1d587703a979601f73222071e16548f628852", type: :room}}, "67540813-dbc4-4dbd-b405-7951c7d11a06" => %Chat.Upload.UploadMetadata{credentials: {<<131, 253, 186, 208, 9, 43, 75, 180, 23, 106, 188, 124, 176, 186, 5, 30, 216, 43, 143, 83, 162, 45, ...>>, <<117, 118, 228, 101, 192, 196, 73, 139, 207, 160, 79, 104, 49, 166, 234, 54, 121, 240, 155, 37, 33, ...>>}, status: :pending, destination: %{pub_key: "0309678f0ca5428d7885a106d1bdd1d587703a979601f73222071e16548f628852", type: :room}}, "69b1d8a3-3524-439d-9993-a8622fd3fa20" => %Chat.Upload.UploadMetadata{credentials: {<<124, 28, 232, 0, 86, 231, 150, 12, 203, 45, 237, 73, 202, 255, 90, 15, 211, 49, 228, 63, 72, ...>>, <<6, 166, 155, 223, 20, 153, 226, 132, 190, 30, 60, 159, 227, 147, 213, 44, 135, 218, 114, 37, ...>>}, status: :pending, destination: %{pub_key: "0309678f0ca5428d7885a106d1bdd1d587703a979601f73222071e16548f628852", type: :room}}, "7248ef85-4e16-4474-9ada-ef6fe6bce55e" => %Chat.Upload.UploadMetadata{credentials: {<<223, 155, 210, 29, 250, 105, 35, 172, 135, 97, 86, 188, 12, 60, 191, 255, 113, 8, 45, 194, ...>>, <<192, 65, 197, 20, 253, 117, 57, 151, 6, 29, 12, 143, 70, 91, 246, 5, 106, 199, 217, ...>>}, status: :pending, destination: %{pub_key: "0309678f0ca5428d7885a106d1bdd1d587703a979601f73222071e16548f628852", type: :room}}, "75baa2f5-dc1b-4b54-861b-39e5156aeb73" => %Chat.Upload.UploadMetadata{credentials: {<<162, 17, 37, 31, 190, 228, 138, 10, 113, 225, 207, 131, 57, 212, 68, 29, 209, 223, 53, ...>>, <<25, 220, 21, 56, 55, 8, 122, 119, 88, 191, 206, 186, 69, 10 (truncated)
```

sviatosluv, 2023-04-24:
@sergey-lukianov I think It's fixed already
https://github.com/Buckitup-chat/chat/blob/ec9f597772c0794d54a0d48fab08b3088184b151/lib/chat_web/live/main_live/layout/message.ex#L343

sviatosluv, 2023-04-24:
@sergey-lukianov isn't it?
https://github.com/Buckitup-chat/chat/pull/266/files

sviatosluv, 2023-04-26:
possibly fixed. let's check in may

closed 2023-05-30 by sergey-lukianov