Buckitup-chat/chat #344: Change Tracker stuck

sergey-lukianov, 2023-05-12:
```
2023-05-12 14:47:49.192000 [warn] setting mirrors: [Chat.Db.InternalDb, Chat.Db.BackupDb]
2023-05-12 14:48:23.570000 [error] GenServer Chat.Db.ChangeTracker.Logger terminating
** (stop) exited in: GenServer.call(Chat.Db.ChangeTracker, :info, 5000)
    ** (EXIT) time out
    (elixir 1.14.1) lib/gen_server.ex:1038: GenServer.call/3
    (chat 0.1.0) lib/chat/db/change_tracker.ex:83: Chat.Db.ChangeTracker.long_expiry_stats/0
    (chat 0.1.0) lib/chat/db/change_tracker/logger.ex:28: Chat.Db.ChangeTracker.Logger.handle_info/2
    (stdlib 4.1.1) gen_server.erl:1123: :gen_server.try_dispatch/4
    (stdlib 4.1.1) gen_server.erl:1200: :gen_server.handle_msg/6
    (stdlib 4.1.1) proc_lib.erl:240: :proc_lib.init_p_do_apply/3
Last message: :check
State: #Reference<0.1555887895.1342701569.16144>
2023-05-12 14:48:37.409000 [error] GenServer Chat.Db.BackupDb.Writer terminating
** (stop) exited in: GenServer.call(Chat.Db.BackupDb.Queue, :demand, 5000)
    ** (EXIT) time out
    (elixir 1.14.1) lib/gen_server.ex:1038: GenServer.call/3
    (chat 0.1.0) lib/chat/db/pipeline/writer.ex:39: Chat.Db.Pipeline.Writer.demand_queue/1
    (chat 0.1.0) lib/chat/db/pipeline/writer_process.ex:45: Chat.Db.Pipeline.WriterProcess.handle_continue/2
    (stdlib 4.1.1) gen_server.erl:1123: :gen_server.try_dispatch/4
    (stdlib 4.1.1) gen_server.erl:865: :gen_server.loop/7
    (stdlib 4.1.1) proc_lib.erl:240: :proc_lib.init_p_do_apply/3
Last message: {:continue, :demand}
State: {:w_state, Chat.Db.BackupDb, "/root/media/backup_db/v.9_files", Chat.Db.BackupDb.Queue, Chat.Db.BackupDb.Compactor, 4, #Reference<0.1555887895.1342701571.104936>}
2023-05-12 14:49:23.581000 [warn] [ChangeTracker stats] 618 awaiting keys, for instance:
```

sergey-lukianov, 2023-07-07:
Think of how to skip change tracker

sergey-lukianov, 2023-07-07:
and get rid of it

buckitupnet, 2024-01-15:
is this one closed?

sergey-lukianov, 2024-01-15:
We still have ChangeTracker working