Buckitup-chat/chat #421: Chat links

buckitupnet, 2023-07-07:
Create links for direct access of the private chats. Like the direct message links but simpler, since nothing additional needs to be stored in db. 
Share this links in form of a qr-code or clipboard from the landing

sviatosluv, 2023-07-19:
@buckitupnet 
What means the chats here? Private rooms or dialogs? Or both?

buckitupnet, 2023-07-19:
Dialogs

sviatosluv, 2023-07-20:
@buckitupnet 
And what does it mean the chat link in this case: the link of specific chat of  two specific users or the link that leads me(the clicker) to the chat with specific user?

More detailed here.
**The first one case.** There is a chat between `Bob` and `Alice`. The chat has a specific and unique link. Only `Bob` and `Alice` have access to the chat by the link. `Peter` can't get into the chat.

**The second one case.** There is the only link of connection with `Bob`. When `Alice` goes by the link - she gets into specific chat between `Bob` and `Alice`. When `Peter` goes by the link - he gets into the chat between `Bob` and `Peter`.

buckitupnet, 2023-07-20:
yes, - exactly, we need the second case.  @sviatosluv actually proposed it, and may have comments or better explanation )

closed 2023-08-03 by sviatosluv